FrankBoesing/FastCRC #15: support buffers larger than 64K

manitou48, 2018-02-07:
Not real critical, but I noticed I couldn't do CRC on buffer larger than 64KB, because benchmark and library uses uint16_t for datalen.  It will be a bit messy to change all of those to uint32_t 

whatever.

GitMoDu, 2018-03-01:
The 8 bit micros really don't like to handle uint32_t, maybe it's something that should be done with a #define?

FrankBoesing, 2019-01-12:
sizeof(int) could help to indentify 8 bit micros.
Manitou, can you add a pull request ?

bxparks, 2020-12-08:
Drive-by comment: On 8-bit AVR processors, `sizeof(size_t) == 2`. So it's probably best to break up the long array into chunks smaller than 64kB and call the `xxx_upd()` functions multiple times.

FrankBoesing, 2020-12-08:
Or just drop 8 Bit support..

bxparks, 2020-12-08:
Totally fair, it's your library.

Since `sizeof(size_t)` cannot be used in a `#if` macro, you can probably do something like this:

```c++
#if defined(ARDUINO_ARCH_AVR) || defined(ARDUINO_ARCH_MEGAAVR)
  #error 8-bit processors not supported
#endif
```

I don't know what other 8-bit processors are out there supporting the Arduino framework.