FrankBoesing/FastCRC #23: Wrong checksum by 1 - about 2s Complement to checksum

dbartelmus, 2019-12-15:
Hello, I've found that my calculated checksum (needed for machine) is "2s Complement", so your script is not 100% accurate without it - can we easily fix it? And do you know how to?
Your script checksum value: 0x3B
Expected checksum value: 0x3A - 2s Complement

FrankBoesing, 2019-12-15:
All checksums are verified, this library is used by many users. This is the first time I've heard of such an error. Unfortunately you don't even say which algorithm you are using. I therefore assume that the error is on your side and that you should convert the result for your purposes. Until I have more information and proof of the error, I won't adjust anything. 
If you have any proof, you are welcome to post it here.
Until then, I will close this issue.

closed 2019-12-15 by FrankBoesing

dbartelmus, 2019-12-15:
Sorry, I was related to CRC8 and some online calculators gives me different results, ex: A2 10 1A 05 62

value: CD - https://www.scadacore.com/tools/programming-calculators/online-checksum-calculator/
CheckSum8 2s Complement 
0x100 - Sum Of Bytes

value: 8D - https://tomeko.net/online_tools/crc8.php?lang=pl

This not happen every time, that's why I asked you about 2s Complement in checksum value. As far as I read about it, initial value should be converted to BINary, then inverted ~ and then subtract by 1.
I'm not any guru so correct me if I'm wrong.

FrankBoesing, 2019-12-15:
Not sure what you want..
I checked your 2nd link, and entered 31 32 33 34 35 36 37 38 39 with init 0x00.
It gives the Result 0xA1, which is the same as with this library
MAXIM (poly=0x31 init=0x00 refin=true refout=true xorout=0x00 check=0xa1) 
..and which exactly is the result of the refernce, "reveng" : http://reveng.sourceforge.net/crc-catalogue/1-15.htm

FrankBoesing, 2019-12-15:
If you need an other result, you may have to convert it or add your algorithm (Pullrequests welcome).

dbartelmus, 2019-12-15:
Thank you for your answer.