FrankBoesing/FastCRC #25: no longer available through the Arduino Library Manager

bxparks, 2020-11-28:
Thanks for this library. My recollection was that this library used to be available through the Arduino Library Manager. But I am no longer able to find this library in the IDE or on the full list of Arduino libraries (https://www.arduinolibraries.info/libraries). Searching for "FastCRC" returns nothing. I'm curious, was this a deliberate change?

bxparks, 2020-12-08:
No response, maintainer likely busy, so I'm going to close this to remove it from my GitHub activities feed.

FYI, if anyone else runs into this problem, the https://github.com/hideakitai/CRCx project has vendored all of FastCRC into its repo, so you can access FastCRC through the thin facade of that project. (It's not clear to me what value that facade provides, but it let's you have access to FastCRC code through the Library Manager.)

Secondly, I created my https://github.com/bxparks/AceCRC problem to solve the one problem I had with this library: By default, it uses large tables with 1024 elements in the lookup table for CRC32, which means that the lookup table consumes 4096 bytes of flash memory on an 8-bit AVR. I created AceCRC to allow me the choice of implementations for CRC32: one consuming 192 bytes of flash, another 220 bytes of flash, or a large one that consumes 1106 bytes of flash. No disrespect against this library, I just couldn't afford 4kB or even 1kB of flash on an AVR just to calculate a CRC32.

FrankBoesing, 2020-12-08:
I saw that it is not available anymore.
That's ok.

bxparks, 2020-12-08:
Sure that's fine, but did you mean to reopen this issue? I'm going to close it again.

FrankBoesing, 2020-12-08:
Yes, others can find it whren it is open ;)

bxparks, 2020-12-08:
Ok, now I just have to 14 days for it to disappear from my GitHub feed... :-)

FrankBoesing, 2021-10-10:
Have changed my mind. It's in the Library Manager now.

closed 2021-10-10 by FrankBoesing