FrankBoesing/FastCRC #28: Any way of doing multiple simultaneous calculations on Kinetis/Teensy?

noisymime, 2022-02-06:
Firstly a huge thank you for the library! I've been using it for some time now and it works really well on both Teensy and AVR based boards. 

I've come across a scenario where I need to have multiple CRC calculations happening in parallel as they each await data to come in over serial. I can create 2 instances of `FastCRC32` and continually call `crc32_upd()` which works fine on AVR, but on Teensy boards it will fail due to it trying to use the same hardware for both calculations. 

As far as I can see there is no way of forcing software calculation on Teensy boards due to the whole sw file being wrapped in a `#if !defined(KINETISK)` check?

I understand this is a fairly niche requirement, but I thought I'd check if there's anything I'm missing that might allow this to work. 
Thanks again for all your work on this!

FrankBoesing, 2022-02-07:
Indeed this is an issue.

FrankBoesing, 2022-02-07:
I thought about this  -  I won't fix that and do not work on Teensy issues anymore.

Sorry :)

closed 2022-02-07 by FrankBoesing