FrankBoesing/FastCRC #29: seed is not initialized if only calling updates

gunderbolt, 2022-02-07:
Hi,

I had a use case where I was iterating through data in a loop and wanted to start by using the update method immediately. This produced incorrect CRC values because the member variable `seed` wasn't initialized. See the following simple example using the kermit CRC16 model:

```
FastCRC16 CRC16;
unsigned short crc = 0x0000;

for (int i=0; i<data_len; i++)
{
    byte data = get_next_byte();
    crc = CRC16.kermit_upd(&data, 1);
}
```

In my case, `get_next_byte()` was the equivalent of getting the next byte of data from EEPROM and I didn't want to copy it into ram first. I am working with an Arduino Uno, so it is using the SW implementation.

Given the way you have grouped the different models into classes by CRC length, I don't see an obvious way to address this. Here are some suggestions:

* Separate CRC models into their own classes and initialize seed in the constructor.
* Add a member variable to keep track of whether it has been initialized or not.
* At a minimum, add a note to the readme indicating that the model method must be called before its `*_upd` method.

As a workaround, I could call the `kermit` method on the first byte and continue iterating through the rest with the `kermit_upd` method, but that seems unnecessarily messy.

Thanks for taking a look at this.

FrankBoesing, 2022-02-07:
Of course the seed is not initialized. Otherwise upd would not work... :)
Here is how to use it: https://github.com/FrankBoesing/FastCRC/blob/master/examples/FastCRC_cont/FastCRC_cont.ino

closed 2022-02-07 by FrankBoesing

gunderbolt, 2022-02-07:
Sure, but that's not documented, nor is it obvious without looking into the code.

My use case is perfectly valid and I'm sure others will come across it too, receiving inconsistent results. For their sake, I hope you'll reconsider.

FrankBoesing, 2022-02-07:
Hm, it exists for six years now. You are the first :) You're also the first who sees a documentation problem.

Just do something like this:

```
byte data = get_next_byte();
crc = CRC16.kermit_upd(&data, 1);

for (int i=1; i<data_len; i++)
{
    data = get_next_byte();
    crc = CRC16.kermit_upd(&data, 1);
}
```

gunderbolt, 2022-02-08:
Yes, I suppose that is a good metric :-)

And the workaround isn't really all that messy.

I guess the only other idea to do this without breaking the current interface would be to add an initialization method for each model that would set the seed. It would really just separate the main model method (e.g.: `kermit`) into two parts (a call to `kermit_init` followed by `kermit_upd`).

And while we're at it, we can add a `kermit_finalize` method so I don't need an extra `crc` variable ;-)

I don't know if all that is worth it. Just some ideas at this point!

Nick